Bodmer/TFT_HX8357 #1: Landscape text is mirrored

DouglasPearless, 2016-01-13:
When running the various tests the txt in portrait mode is correct, but in landscape it is mirrored, for example the "graphics test_240x320" code.
[image]

DouglasPearless, 2016-01-13:
I have narrowed down the issue, it seems that only "myGLCD.setRotation(1);" produces the correct result, and all other rotations either flip or invert the text.

Unless I am missing something when reading the code?

Bodmer, 2016-01-13:
Hi Douglas, Thank you for taking the time to report this issue.
I have run this test on my displays bought from Banggood and they work fine in all orientations.
Looking at the nice picture you posted I see there are some subtle visual differences between yours and mine in the area of the connections from the display to the shield PCB. Can you tell me where you got the display from?

There is a set of registers that controls mirroring and so this could be corrected on your display. I will think on this further and see I can add an option to the User_Setup file so these differences between the displays can be handled.

Bodmer, 2016-01-13:
Can you try four other rotation values of 4, 5, 6 and 7 and tell me which ones display correctly. For example:

myGLCD.setRotation(4);

You should find that three out of these four values looks OK.

DouglasPearless, 2016-01-13:
I tried those but they were mirror images.  I am away from my stuff for another 6 hours and will post photos of rotation 1-7

Bodmer, 2016-01-14:
OK, it would be best to use the example "graphicstest_320x480" as the one called "graphicstest_240x320" tries to force the display area to be smaller and this may be causing problems.

There are quite a few bits in a register called MADCTL that flip, rotate and mirrors the display. Can you locate the following line in the library HX8357.cpp file:

#define MADCTL_BGR 0x08

Change it to read:

#define MADCTL_BGR 0x0A

Then try rotations 0 - 7 and see is any look right. This tweak changes a bit that does a horizontal flip that might help.

There are two other settings that would flip different axes:

#define MADCTL_BGR 0x0B
and
#define MADCTL_BGR 0x09

These also change a bit called GS which flips a different Axis.

If you try each in turn with rotations 0-7 then one of the settings should give the right display.
If you let me know which ones work then I can add an option to the library to correct it for others.

It is a bit odd that the library works fine with my display. It may be that your display has a different controller chip that has a similar instruction set.  Can you give me a link to where you bought the display from then I can investigate further.

Bodmer, 2016-01-14:
I had a look on Banggood and some fold are complaining that they have received a display fitted with the ILI9481 controller, so that is probably why the display is flipped. Forthunately the register set is very similar so we should be able to get your display running with this library. I will add an option in User_Setup to select a ILI9481 configuration once we have figured out the correct settings.

Thanks.

DouglasPearless, 2016-01-14:
This is the one that I have http://www.aliexpress.com/item/Free-shipping-3-2-inch-TFT-LCD-screen-module-Ultra-HD-320X480-for-Arduino-MEGA-2560/32283720984.html

DouglasPearless, 2016-01-14:
Here it is with #define MADCTL_BGR 0x0A

#define MADCTL_BGR 0x0B

#define MADCTL_BGR 0x09

Cheers

DouglasPearless, 2016-01-14:
As my Pics did not upload, I modified the sketch and code to try all possible MADCTL values.

Here is a link to a youtube video that shows the MADCTL varying from 0 to 255 so this may help you figure out how this controller works!

https://www.youtube.com/watch?v=MZ0mbfq6mgw

DouglasPearless, 2016-01-14:
Note: Some of the text change colour too!

Bodmer, 2016-01-14:
Thanks, the colour change is to be expected as 1 bit in the control word changes the colour order between RGB and BGR.

Using you video I have worked out the bit pattern you display needs and added an option called MIRROR to the User_Setup.h file. Enabling this by removing the // should correct the mirroring on your display. Interestingly my HX8357 behaves the same whether MIRROR is enabled or not so this is a good indicator your display has an ILI9481 controller fitted, in other respects that controller is compatible with the HX8357B.

The GitHub repository has been updated so please download this new copy of the library, enable MIRROR and let me know if all is now OK.

Thanks.

DouglasPearless, 2016-01-14:
Now works perfectly!!!

I wonder is there a way for your code to determine which controller it has and to automatically set the MIRROR variable?

Bodmer, 2016-01-14:
Excellent!
The display itself has a read line but this is not wired up on the shield PCB so sadly the controller ID registers cannot be read by the Mega. Otherwise as you say the selection could be automatic.

closed 2016-01-14 by Bodmer