Bodmer/TFT_HX8357 #3: can't seem to get DrawString to work consistantly

alfajunior105, 2016-03-22:
Hi, below is some sample code , I cant seem to figure out why
the first works ok but the second one doesn't display at all?
  i.e. it doesnt display the "ABCDEF " ,
what am i doing wrong?

#include <TFT_HX8357.h> // Hardware-specific library
#include "Free_Fonts.h" // Include the header file attached to this sketch
#include <stdio.h>
#include <SPI.h>
TFT_HX8357 tft = TFT_HX8357();       // Invoke custom library
const int chipSelect = 53;

void setup(void)
{
  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK); // Clear screen to navy background
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  Serial.begin(9600);
}

void loop()
{
    tft.setTextColor(TFT_WHITE, TFT_BLACK);
    tft.setTextDatum(TC_DATUM); 
    tft.drawString("1234567890123456" , 5, 55, 6);
    tft.drawString("ABCDEF " , 5, 99, 6);
    delay(4000);
}

Bodmer, 2016-03-22:
The library font set was originally developed for the UNO which does not have enough memory for a complete font set.  Font 6 therefore only contains numerical characters and a few others to support time and drawing floating point numbers:
Thus the font only contains characters [space] 0 1 2 3 4 5 6 7 8 9 : - . a p m
If you want to use other characters you will have to use one of the Free Fonts.

alfajunior105, 2016-03-23:
thanks foryuour help - I was pulling my hair out :-)
 I use a free font then
Thanks

closed 2016-03-24 by Bodmer