Bodmer/TFT_HX8357 #4: Bigger font

rafik73github, 2016-07-11:
Hi
How to create bigger font, size 72, 96 or other? Arial or Times New Roman, maybe Serif.
Greetings

Bodmer, 2016-07-11:
The library supports the Adafruit free font format. Linux tools for this are in the fontconvert folder.  I have not used these tools myself, they were created by Adafruit.
The library also includes Run Length ~Encoded fonts but these are difficult to create so it is best to stick with the Adafruit freefont style.

rafik73github, 2016-07-11:
OK. Display the digit timer when using the enlarged font (tft.setTextFont(6); tft.setTextSize(1);), the text blinks. Is it normal?

rafik73github, 2016-07-11:
This is my code:

#include <TFT_HX8357.h> // Hardware-specific library
TFT_HX8357 tft = TFT_HX8357();       // Invoke custom library

int m=0;
int s=45;
int timerstat = 1;

void setup()
{
  
// Setup the LCD
tft.init();
  tft.setRotation(1);
tft.fillScreen(TFT_BLACK); // Clear screen to black background
tft.setTextColor(TFT_GREEN,TFT_BLACK); //green
tft.setTextFont(8); 
tft.setTextSize(2);
}

void timer()
{ 
if(m == 0 && s == 0) {timerstat = 0;}
if(timerstat == 0) //----------------------------------uptime
{
  tft.setTextColor(TFT_RED,TFT_BLACK); //red
 //delay(225);
 s++;  
 if(s > 59) 
 {
 m++;
 s=0;
 } 
}
else  //---------------------------------------------downtime
{
if(m == 0 && s < 31) // blink when 30 sec left...
{
  
  tft.setTextColor(TFT_YELLOW,TFT_BLACK); //yellow
  
 //delay(225);
 
 s--; 
 if(s < 0) 
 {
 m--;
 s=59;
 } 
 
}
else
{
  tft.setTextColor(TFT_GREEN,TFT_BLACK); //green
//delay(225);
 s--; 
 if(s < 0) 
 {
 m--;
 s=59;
 }
}
} 
} // end void timer

void timerDisplay()
{
 int m1 , m2 ,s1 , s2;
if (m > 9) {
    m1 = (m - (m % 10)) / 10;
     } else {
    m1 = 0;
  }  
    m2 = m % 10;
    
    if (s > 9) {
    s1 = (s - (s % 10)) / 10;
     } else {
    s1 = 0;
  }  
    s2 = s % 10;
   
  tft.drawNumber(m1,5,70,8);
 
  tft.drawNumber(m2,113,70,8);

  tft.drawNumber(s1,253,70,8);
 
  tft.drawNumber(s2,366,70,8);
 
} // end void timerDisplay

void loop()
{
timerDisplay();
timer(); 
delay(1000);
//tft.fillScreen(TFT_BLACK);
}

Bodmer, 2016-07-13:
Yes, it is normal when a scaled font (x2 in this case) is used as there are so many pixels to draw and the character is blanked first.

The effect can be made less visually intrusive by only drawing characters when they change. For example declare global variables:

int om1 = 99, om2 = 99,os1 = 99;

Then is the code use a conditional test like this:
s2 = s % 10;
if (m1 != om1) tft.drawNumber(m1,5,70,8);
om1 = m1;
if (m2 != om2) tft.drawNumber(m2,113,70,8);
om2 = m2;
if (s1 != os1) tft.drawNumber(s1,253,70,8);
os1 = s1;
tft.drawNumber(s2,366,70,8);

You will also have to add conditional code when the colour changes too using the same approach, but these changes will let you see the improvement.

rafik73github, 2016-07-14:
Thank you for your help. I tried to convert large font, but I can not deal with it. I do this on Fedora 23. I used the description on this page:
https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts.
But it does not work. I installed GNU and Freetype, type a command fontconvert parameters and displays a "command not found".

rafik73github, 2016-07-14:
How to display a bitmap? I try this:
http://javl.github.io/image2cpp/
and my code:
const unsigned char myBitmap [] PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x0f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xff, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x7f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0xff, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 
0xff, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 
0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xff, 0xf0, 
0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xf0, 0x00, 
0x00, 0x00, 0x00, 0x07, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0x9f, 0xf0, 0x00, 0x00, 
0x00, 0x00, 0x0f, 0xf9, 0xff, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x1f, 0xf0, 0x00, 0x00, 0x00, 
0x00, 0x3f, 0xe1, 0xff, 0x00, 0x00, 0x00, 0x00, 0x07, 0xfe, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 
0x7f, 0xc1, 0xff, 0x00, 0x00, 0x00, 0x00, 0x0f, 0xf8, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x01, 0xff, 
0x81, 0xff, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xf0, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x03, 0xfe, 0x01, 
0xff, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xc0, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x0f, 0xfc, 0x01, 0xff, 
0x00, 0x00, 0x00, 0x00, 0xff, 0x80, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x1f, 0xf0, 0x01, 0xff, 0x00, 
0x00, 0x00, 0x03, 0xfe, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x7f, 0xe0, 0x01, 0xff, 0x00, 0x00, 
0x00, 0x07, 0xfc, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0xff, 0x80, 0x01, 0xff, 0x00, 0x00, 0x00, 
0x1f, 0xf8, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x03, 0xff, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x3f, 
0xe0, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x07, 0xfc, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0xff, 0xc0, 
0x00, 0x1f, 0xf0, 0x00, 0x00, 0x1f, 0xf8, 0x00, 0x01, 0xff, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 
0x1f, 0xf0, 0x00, 0x00, 0x3f, 0xf0, 0x00, 0x01, 0xff, 0x00, 0x00, 0x07, 0xfe, 0x00, 0x00, 0x1f, 
0xf0, 0x00, 0x00, 0xff, 0xc0, 0x00, 0x01, 0xff, 0x00, 0x00, 0x0f, 0xf8, 0x00, 0x00, 0x1f, 0xf0, 
0x00, 0x01, 0xff, 0x80, 0x00, 0x01, 0xff, 0x00, 0x00, 0x3f, 0xf0, 0x00, 0x00, 0x1f, 0xf0, 0x00, 
0x03, 0xfe, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x07, 
0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x07, 0xff, 
0xff, 0xff, 0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x07, 0xff, 0xff, 
0xff, 0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x07, 0xff, 0xff, 0xff, 
0xff, 0xff, 0xff, 0xe0, 0x7f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xfe, 0x07, 0xff, 0xff, 0xff, 0xff, 
0xff, 0xff, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x1f, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 
0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1f, 0xf0, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
};
void setup()
{
tft.drawBitmap(0,0,myBitmap,60,80,TFT_YELLOW);
}

displays artifacts and not graphics.

source image:
[maybe.zip](https://github.com/Bodmer/TFT_HX8357/files/364420/maybe.zip)

Bodmer, 2016-07-18:
The bitmap function in the library expects the bitmap array to be byte aligned, i.e a multiple of 8 bits wide to be compatible with the Adafruit_GFX library.

If you try this you will see that the bitmap is otherwise OK:

tft.drawBitmap(0,0,myBitmap,120,40,TFT_YELLOW);

120 is divisible by 8 so it works but you get two 4's side by side as each bitmap is actually 60 pixels wide.
You will have to write a new function if you wish the bitmap to be an arbitrary number of bits wide.

rafik73github, 2016-07-18:
Thank you for help.I'll try this.

closed 2016-08-06 by Bodmer