Bodmer/TFT_HX8357 #6: Turn off the display?

Mariusz1, 2016-08-06:
Thanks for the great work :)
Do you have an idea how to turn off the display using code (Arduino Mega2560)?

Bodmer, 2016-08-06:
There are two power consumers on these displays. The main one is the backlight LEDs, these consume the bulk of the power, the second is the display controller which has quite a low consumption even when running.

It is possible to command the display controller into a lower power mode but the power saved will be small.

The back-light LEDs are wired direct to the Arduino's 5V power and thus cannot be switched off.

So you would have to build a display adapter that allows you to cut the power to the LEDs or hack the display PCB, e.g. via a transistor controlled by the Arduino. If you decide to do that then I can write a function for you that switches the display controller to a lower power mode, but as noted above that will save only a very small amount of power compared to the back-light LEDs and thus without a back-light control it is not worth doing for the tiny power saving.

closed 2016-08-06 by Bodmer

Mariusz1, 2016-08-06:
Thanks for the answer.
My display is embedded into a Mega2560 shield thus I am not sure if I can hack only the LEDs (at least I can’t see a way how to do it). However I thought about disconnecting the top two pins from the shield’s connector (both are 5V) and control them externally via a transistor. Of course this approach will turn off the entire display shield. What do you think about this? What could be possible results for the software library, e.g. if the display “disappear” (no power)?

Bodmer, 2016-08-06:
If you remove +5V from the display you will also have to ensure all data pins driving the display are at logic 0 or are switched to inputs, otherwise the data lines will power the display through parasitic diodes in the buffers.

[http://i.imgur.com/qwvqRtF.png](url)

I do not know which display you have, on my 3 different displays the LED current goes through a 2R00 surface mount chip resistor visible on the back of the board. This will likely be written on the resistor in tiny letters and numbers. On one of my boards there is only one resistor R1 and that controls the LED current, on another there are places for three resistors R1-3, and the LED current control resistor is R3. If you soldering skills are up to it you could put in a series transistor, probably a PNP if the resistor is fed from +5V,

Bodmer, 2016-08-06:
In answer to this question:
"What could be possible results for the software library, e.g. if the display “disappear” (no power)?"

Once the display is powered off you must not call and library functions until it is powered again otherwise the data pins will attempt to power the display if any are at logic 1.

When the display is powered again you will have to re-initialise it by calling tft.init() once. This will configure the display and set all the control lines into their correct state again.

Mariusz1, 2016-08-06:
Thank you very much for all the information :)
It looks like I have the "three resistors R1-3" version. AFAIK the code "2R00" means 2 ohms. For me it is a bit strange beacuse for LEDs I would expect about 100R or more?
Your sugestion is to unsolder the resistor and attach a transistor with similar resistor (2 ohms)?

Bodmer, 2016-08-06:
Yes, the 2R0 is low... so I decided to confirm that is it's function, so I unsoldered it on one of my displays and the backlight LEDs are still on!  This is odd as I have a cracked display on one board so I dismanteld it and I looked at the LED feed and that resistor does appear to be associated with the LED lines. But clearly I am wrong as the function of the good display was not changed whether the resistor is present or not.

So my analysis of the circuit is wrong, so iit is best ignore that suggestion and go back to you original suggestion of switching the +5V line.  I can't see any other option.

Mariusz1, 2016-08-06:
OK - thank you very much :)