Bodmer/TFT_HX8357 #7: reference to 'SD' is ambiguous error

lana726, 2016-10-11:
Hi,

I'm trying to run the Draw_SDCard_Bitmap example but I keep getting this error message.
[image]

Bodmer, 2016-10-12:
Thanks for reporting this.  Only one SD library should be used and I see both are called up in the example, I have corrected this on GitHub.

Either use the lines thus at the start of the sketch with the 2 SdFat lines commented out:

// SD Card library, comment out next line if the SdFat library is used
#include <SD.h>

// We can use the SdFat library instead and it will be a bit faster, uncomment next 2 lines
//#include <SdFat.h>           // Use the SdFat library for the Due
//SdFat SD;                    // Permit SD function call compatibility with no sketch changes :-)

Or with the SD.h line commented out like this:

// SD Card library, comment out next line if the SdFat library is used
//#include <SD.h>

// We can use the SdFat library instead and it will be a bit faster, uncomment next 2 lines
#include <SdFat.h>           // Use the SdFat library for the Due
SdFat SD;                    // Permit SD function call compatibility with no sketch changes :-)

Does that fix the error message?

Bodmer, 2016-10-12:
Sorry if you got multiple replies, had trouble getting the code markup syntax right!

lana726, 2016-10-12:
Hi thanks so much for that, I believe that has fixed the problem however I am now getting a new error message. Sorry I'm fairly new to this.
[image]

Bodmer, 2016-10-14:
The compiler is not finding an SD library.  Make sure the following line is not commented out:

#include <SD.h>

Comment out the following 2 lines in the sketch thus:
//#include <SdFat.h>
//SdFat SD;

closed 2016-10-14 by Bodmer