Bodmer/TFT_HX8357 #8: memory overflow issue

RobiGoBH, 2016-12-18:
Hi, 
I'm working on a Mega based boiler controller. The whole code is a bit big and contains a few icons 
When I compile the application I have something like this:

> Sketch uses 121,210 bytes (47%) of program storage space. Maximum is 253,952 bytes.
> Global variables use 4,501 bytes (54%) of dynamic memory, leaving 3,691 bytes for local variables. Maximum is 8,192 bytes.
> 

If I add one more 32x32 icon then I have this:

> Sketch uses 123,230 bytes (48%) of program storage space. Maximum is 253,952 bytes.
> Global variables use 4,493 bytes (54%) of dynamic memory, leaving 3,699 bytes for local variables. Maximum is 8,192 bytes.

When I'm in the second scenario then the tft.drawString instead of drawing a string draw nothing or the full screen looks like randomly filled with hairlines, some icons drawn well but only a few, except that nothing readable... (I saw something similar on my PC when my video card died)
If I remove the last icon from the code to reduce memory footprint then everything working just normal.
I have version 0.27.0 installed.
Have you any idea why is that happening?

Bodmer, 2017-01-15:
Sorry for the delay in replying.

 

This is down to a limitation of the Mega as far as I can see, that happens is that the processor can only access 64Kbytes and thus has to uses memory pages. I guess the original Arduino ICE and bootloader designers did not foresee users having large arrays containing image data which pushes the start of the executable sketch over the 64K boundary.

 

I am sorry but I have tried to get to the bottom of this and failed. It does seem to be bootloader or IDE compiler limited as it works fine on a Due that has more memory and does not need to use the page system.

 

Since your project has got rather large it would be worth swapping to the Due if you can.  What TFT are you using?

 

Regards,

Bodmer

RobiGoBH, 2017-01-17:
No worries, thank you for investigating that issue. I'm using a 3.2 inch 480x320 TFT shield for Arduino Mega (I think it is based on HX8357B controller) 
What is weird, the program itself is working well, so it is doing all the logic, controlling relays, only the text output is drawing random lines instead of characters, and some icons are not showing correctly. 

Kind regards, 
Rob

RobiGoBH, 2017-01-17:
By the way, do you have experience linking the arduino board with the TFT shield using an ribbon cable? I tried that but the result was terrible (the cable length was maybe 10cm)

Bodmer, 2017-01-18:
Ribbon cable can be a problem with high speed signals due to "ground bounce" and stray capacitance cross-talk between wires.  Beefing up (more wires or thicker wire) the GND (0V) connection between the display and the Arduino board might help. The write strobe is the critical signal so you could try cutting that wire away from adjacent ones or using a separate discrete wire.

closed 2017-04-23 by Bodmer