Bodmer/TFT_HX8357 #9: Bitmap read of SD CARd

izabdi, 2017-02-03:
Hi, I downloaded the library and i was particuarly interested in the example where  you read the sd CArd to load  a bitmap mage. i tried it and it had an error..pointing the the bitmapfunctions tab saying FILE WAS NOT DECLARED IN THE SCOPE... Can you help me with this error.. iv tried moving the libraries in the same work area.. my teachers havnet been able to figure it out. Working on  a project and im really interestd on displaying images. Another thing, I was playing around with the code, I commented the function for DRAMBMP and left the DRAWRAW.. it worked. But my problem there is that i cant convert my picts to raw. the converter UTFT library has doesnt work. im using a 3.2 inch 480x320 hx8357 shield for mega. Cn you help me with eithr the code or the raw conversion?

Bodmer, 2017-02-04:
Hi,
Are you using a Mega or a Due?
Which version of the Arduino IDE are you using?
What is the problem with the UTFT converter, is it producing the raw file? If yes, then what happens when the raw file is drawn on the TFT, does an image appear but the colours are wrong?

Bodmer

Bodmer, 2017-02-04:
I have tried the sketch and it works fine, if I comment out the #include <SD.h> I get the error message you mention.

Check you are including the SD library and that it is present in the Arduino IDE. 

If you are trying to use the SdFat library then that must be loaded in the IDE.

As per comments in the sketch, only use SD or SdFat, do not try to include them both.

izabdi, 2017-02-05:
hmm no i have the inclde sd.h uncommentd...i get that error still... and well in regards to my other question, wen u conver to a raw file... i tried converting images and it doesnt show up...can u guide me?

Bodmer, 2017-02-06:
In the Arduino IDE, click the File>Preferences menu option and tick the box "Show verbose output during compilation" in the pop up window, click OK that at bottom right of that window.

Now click the compile button at to left (a tick symbol).

Use the "Copy error messages" button at bottom right side of IDE window.  Paste the error message into a comment box here on Github so I can see what the compilation error messages say.

izabdi, 2017-02-07:
Arduino: 1.0.6 (Windows NT (unknown)), Board: "Arduino Uno"
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=106 -IC:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\variants\standard -IC:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master -IC:\Program Files (x86)\Arduino\libraries\SD C:\Users\ACER-PC\AppData\Local\Temp\build3725313147770010465.tmp\Draw_SDCard_Bitmap.cpp -o C:\Users\ACER-PC\AppData\Local\Temp\build3725313147770010465.tmp\Draw_SDCard_Bitmap.cpp.o 

In file included from Draw_SDCard_Bitmap.ino:11:
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/TFT_HX8357.h:124:1: warning: "HX8357B" redefined
In file included from C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/TFT_HX8357.h:19,
                 from Draw_SDCard_Bitmap.ino:11:
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/User_Setup.h:13:1: warning: this is the location of the previous definition
In file included from C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/TFT_HX8357.h:29,
                 from Draw_SDCard_Bitmap.ino:11:
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/glcdfont.c:11: warning: only initialized variables can be placed into program memory area
In file included from C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.h:1,
                 from C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/TFT_HX8357.h:33,
                 from Draw_SDCard_Bitmap.ino:11:
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:8: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:26: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:32: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:38: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:44: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:50: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:56: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:62: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:68: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:74: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:80: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:86: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:92: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:98: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:104: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:110: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:116: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:122: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:128: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:134: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:140: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:146: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:152: warning: only initialized variables can be placed into program memory area
C:\Program Files (x86)\Arduino\libraries\TFT_HX8357-master/Fonts/Font16.c:158: warning: only initialized variables can be placed into program memory area
[... cut ...]
bitmap_functions.ino:84: warning: unused variable 'row'
bitmap_functions.ino: In function 'uint16_t read16(File&)':
bitmap_functions:195: error: 'uint16_t read16(File&)' redeclared as different kind of symbol
Draw_SDCard_Bitmap:17: error: previous declaration of 'uint16_t read16'
bitmap_functions.ino: In function 'uint32_t read32(File&)':
bitmap_functions:202: error: 'uint32_t read32(File&)' redeclared as different kind of symbol
Draw_SDCard_Bitmap:18: error: previous declaration of 'uint32_t read32'

Bodmer, 2017-02-07:
OK, the first line is  Arduino: **1.0.6** (Windows NT (unknown)), Board: **"Arduino Uno"**

1. 1.0.6 is a really really old version of the Arduino IDE, use the latest 1.8.1 version.
2. The sketch is compatible with a Mega, not an UNO

I think the error wil go away if you make both these changes. Report back how you get on.

izabdi, 2017-02-07:
oh yea.. ill do the first one...and number two well im working on other projects using the uno but it shows the same errors

closed 2017-04-23 by Bodmer