Bodmer/TFT_HX8357 #14: strange pixel during on the display during running on Atmega2560

mobifu1, 2017-10-20:
you can load my using code for testing:
https://github.com/mobifu1/arduino_pa_controller

After 1 minute of running, the software produce many strange colored pixel.
The pixel position on the display looks like a random process.
maybe you can reproduce the error.
regards Mobifu1

Bodmer, 2017-10-20:
I suspect you are using the HX8357B controller. This version of the controller appears to have a silicon level bug in the controller that generates spurious pixels. I extect that is why there is a HX8357C controller as that one does not exhibit the same problem.

Unfortunately there is nothing I can do to fix this (I have spent a long time investigating this!) except suggest that the compete screen is drawn occasionally to wipe out the odd pixels.

mobifu1, 2017-10-21:
Hello Bodmer,
thank you for the information and your work, yes the board was cheap, thats the problem :-)
regards from Hamburg Mobifu1

 

 

aajahnke@aol.com

Bodmer, 2017-10-21:
The 3" HX8357C based display is also cheap for it's size and works really well. I will update the ReadMe to notify users of this issue. Thanks.

closed 2017-10-21 by Bodmer