Bodmer/TFT_HX8357 #30: Arduino Due

11Naweed11, 2020-01-09:
Can you modify the library to support Arduino Due? Because mega is just too slow.

Bodmer, 2020-01-09:
I created a[ Due library here](https://github.com/Bodmer/TFT_HX8357_Due) sometime ago. Try that one.

11Naweed11, 2020-01-09:
Thanks, haven't tried anything but at least it compiled without any errors .
one more thing, can you please point me out to a guide about wiring ? just in case I don't connect the module incorrectly with due.

11Naweed11, 2020-01-09:
and also the library only says ili9481, I have ili9486 would it make a big difference ?

Bodmer, 2020-01-09:
There is no writing guide but the library has quite a few examples and is, at the basic function level, compatible with Adafruit_GFX so you can use the Adafruit online tutorials for the basics.

The ILI9486 is not supported by the Due variant of the library. As the Due variant is based on this Mega version you could port over the differences. The Due library might work with the ILI9486 with the ILI9481 option selected, you could try that on this Mega version before you commit to a library change.

All my "Arduino" effort these days goes into my TFT_eSPI library and that keep me very busy in my spare time so I do not plan to work on these older libraries anymore. The TFT_eSPI library is being updated to support the STM32 Nucleo 64/144 "UNO" compatible boards which are available with much higher performance.

Depending on what you want to do... or expect should be possible... you might also find the Due still does not provide the performance you are looking for.

11Naweed11, 2020-01-09:
Thanks alot for your time. I meant wiring, interfacing the module with the due. In other words which pin of LCD goes to which pin of due?
In case of performance, I used mcufriend LCD shield with driver ic LGDP4532 with due and noticed significant performance changes over uno or mega.

Bodmer, 2020-01-09:
There is no wiring, you just plug it in, the library is for [16 bit displays like this one](https://www.aliexpress.com/item/32917044135.html?spm=a2g0o.productlist.0.0.645d3b6dWsEKBu&algo_pvid=4f75599b-c039-42bf-a28e-ff259a56feaf&algo_expid=4f75599b-c039-42bf-a28e-ff259a56feaf-0&btsid=c1c78195-5470-4e7e-a917-eb02ceba069f&ws_ab_test=searchweb0_0,searchweb201602_2,searchweb201603_53).

11Naweed11, 2020-01-09:
well, mine is a bit different. its not a shield. 
[image]

11Naweed11, 2020-01-09:
I found this in the library .cpp file but I don't understand what is A, B or C ? are they port?

[image]

Bodmer, 2020-01-09:
That display will be a bit of a nightmare to wire up, you can get special adapter shields for it to avoid wiring and these have level translators but you need to ask on the Arduino "Display" forum. It will not work with a Mega without a shield as there are no 5V to 3.3V level translators on the display board and the display expects 3.3V logic. It might work with a Due.

Yes the letters are the processor port designations. Ask for help on the Arduino forum.

closed 2020-01-09 by Bodmer