Bodmer/TFT_HX8357 #34: White screen

msowa2424, 2020-02-09:
MEGA 2560 board with 3.5" TFT LCD Shield driver ili9481 no touch screen. User_setup.h uncommented for #define ILI9481. Sketch uploading successfully. Serial monitor on and displaying test data but screen is white all the time. White screen also for all other examples.

⸮HX8357 Test!
Benchmark                Time (microseconds)
Screen fill              212728
Text                     41364
Lines                    214860
Horiz/Vert Lines         18404
Rectangles (outline)     11408
Rectangles (filled)      538204
Circles (filled)         155972
Circles (outline)        189296
Triangles (outline)      49084
Triangles (filled)       272252
Rounded rects (outline)  62488
Rounded rects (filled)   606004
Done!

Bodmer, 2020-02-09:
Try a different driver. Your screen may not be an ILI9481 type.
Post a link to the sellers web page. Post a picture of the back of the TFT board.

Fallen2504, 2020-02-10:
Hey I am having same problem. i am using 3.2 inch ili9481. I ran the graphic test without editing setup file( because i could not find it . It was not in library folder) and am getting same results as you.

Fallen2504, 2020-02-10:
I am also having the same problem. I am using a 3.2 inch ili9481 with arduino mega. When i run graphic test lcd just turns up white and serial monitor shows same results.

johng1uvy, 2022-08-16:
Hi I have the same fault as above.
The screen works great with a Mega 2560, but not with a Due using the same TFT all I get is a white screen.
Any help please.

Bodmer, 2022-08-16:
This library is for Mega only. The Due version is here:

https://github.com/Bodmer/TFT_HX8357_Due

johng1uvy, 2022-08-19:
Hi sorry posted on the wrong section. It is the library for the Due that I tried but.
The screen works great with a Mega 2560, but not with a Due using the same TFT all I get is a white screen.
Any help please.