Bodmer/TFT_HX8357 #39: Screen gray with this library for my ILLI 9481

LukeDaniel16, 2020-04-11:
I tried to use your library with my screen but it only stays with the gray screen, in the examples I used, currently I use the library MCU_Friend_kbv, but it is very slow and I want to use yours, this is my tft: [https://www.youtube.com/watch?v=R_V_lzAbnb8](video demonstration of use)

LukeDaniel16, 2020-04-11:
[https://lh3.googleusercontent.com/-giA7zUzuxJw902b0ljdOB3hApIyaMnVZfubgHjJzBfrq-jkCBJuRdOf-mBXP_iX3pvkUV-FrxWVLVefodY8dv41KQAXM0xwVKug_UPbyrkybNobQOiI1mMgWU0YgF9gRg_EEC4sbYnO5nWcHhSCwNGGc1wP1EO2elUqEcHa7iiJGfL9FXeKtFnQiJg6ufX6-xi5KBPUiz-4zbZPbcwBCcuj6toEgNz5FlxWIuNIYWvdxb3hsWKcdaAh8V-vI85tKS4500pn5yMBYfc4QCOomJC0B4_eOm15KFci7OpYR9DBQWnXPVBa4TTixN01G25D3KanYVXYFqchxyTCrtZowy8usOKBTSLL8FWmZs3L1vh9T5uasqC4RjEpy7xiHulVj47h3Hp-BdP64ACEchOtb0fC76zD3uHP2LD8E1_Ix-D6rhKSibgDzErj0PhD6UvJMMGwZ6KBa7-E75WBbsSe2xts4QXNyaB7BecbsJFjoLd81GePclUN08-mrlCuzD_6jxNV0FyyViEaiND0Z5wdE1-lzp4EVRyuf8rviWocxXD0iNWngUmcxzKNOzCgCo18ZY7__j2J9XFGfaSt0ibCJA_3xYxPeyPgptIJypHjNQc5Ps7ZEiG5Om3LVQdD3qpUHbkYjarrsnn9YbPGvCapmYqf5u58oFC_1KgNu_SaZ6blNiRGmUX9apEcF-acJQUqGTioUcXTPvLag4eeqEHLxIPJeQ=w843-h632-no](image of circuit )

1martinlipka, 2020-09-08:
Try hacking out the power command configurations : removing power setting '0xC3' worked for me.

try comment out these lines, and recompile.     
    
    writecommand(0xC2);        // Power Control 3
    writedata(0x00);