Bodmer/TFT_HX8357 #45: Adjust the brightness

Palmnarate, 2021-01-08:
Hi Bodmer,

I use the Mega with HX8357B with your lib.
So how can I adjust the brightness?

Thank you,
Palm

Bodmer, 2021-01-08:
The display is illuminated by white LEDs so I assume you wish to change the LED brightness.

There is no built in capability on the boards that I have, but yours may be different. Usually the LEDs are connected directly to the 3.3V supply rail on the board. This 3.3V is produced by a regulator on the board that is powered from 5V from the Arduino pins.

I expect you will need to modify the board and fit a transistor switch circuit, then assign a pin on the Mega to generate the PWM signal. Modifying the board will involve some delicate soldering as the a resistor (typically 2R2 i.e 2.2 Ohms) may need to be removed and perhaps some circuit board tracks cut. I am not able to advise how this can be done but you may find information on the internet on how to control an LED's brightness with an Arduino and be able to work out what modifications are needed to the display board.

Of course fitting some darkened glass/ plastic infront of the screen is another way to get the brightnes down.

closed 2021-01-08 by Bodmer

Palmnarate, 2021-01-11:
Thank you so much.

I will check for this solution.

The point is when I open the tft display for a long time ex. 2-3 years I worry about lifetime of the display.
So I think if I turn off the display or reduce the brightness when not use it will improve the display lifetime.
or do you have any other solution?