Bodmer/TFT_HX8357 #49: Bitmap Image Not Working

piggyking21, 2022-08-08:
Hi, 
I have been trying to use this library to display simple .bmp images on a Arduino mega2560 shield I bought a while ago. Using any examples for the display work perfectly fine but when I loaded what I think is a correctly formatted bitmap image using the "Draw_SDCard_Bitmap" example code, I get some sort of corrupted image on the display. Its most likely my fault but any help would be appreciated. For reference I'll include the result I get on the shield and my original image. I'm not very experienced with Arduino related projects so if I happened to have made a stupid mistake, sorry in advance :) 

Here is the image I'm trying to use (not formatted for 480x320 or in a .bmp)
[](https://imgur.com/io5pTzq)

and here is what the screen shows (you can kind of see that the image is there but sort of corrupted, sorry for the horrible lighting)
[](https://imgur.com/1eZrOqt)

sorry if this is not so much an issue with the library (I just didn't know where to ask for help)

Bodmer, 2022-08-08:
Post the image as a zip file, just drag and drop.

It may be an endian issue, tft.setByteSwap(true) in setup() may fix it. Also try false.

Bodmer, 2022-08-08:
The bmp file may be in the wrong format. Post a zip file of the image, just drag and drop.

piggyking21, 2022-08-08:
[image.zip](https://github.com/Bodmer/TFT_HX8357/files/9283503/image.zip)

Ok, here is the png of the image I have on the SD card 
also I'll include a working picture of what the screen showed me.

[image]

Bodmer, 2022-08-08:
Post the original .bmp image as the format may be incompatible with the library and I can check it.

piggyking21, 2022-08-08:
[DOT.zip](https://github.com/Bodmer/TFT_HX8357/files/9284717/DOT.zip)